RickMcConney/PenPlotter #1: Error running from source

patriciorivera, 2015-08-29:
Hi. Great Job!! Im getting this error running v0.5 from source processing 2.2.1. on mac OS. Thanks!

[image]

RickMcConney, 2015-08-29:
You may be running against a different version of the controlP5 library. I have uploaded the version of the library I am using to the V05 release, Try downloading the libraries.zip file and unzipping it in the processing/libraries directory.

RickMcConney, 2015-08-29:
Leave open until fix confirmed

patriciorivera, 2015-08-29:
I confirm that work fine only with controlP5 version 2.2.5. By the way, I have tested versions 0.5.4, 1.5.2 and 2.0.4, failing in all cases.

closed 2015-09-01 by RickMcConney