RickMcConney/PenPlotter #2: gondola out of frame if not set home at the beginning.

patriciorivera, 2015-09-03:
If not set home previously, pressing arrows or moving the gondola at the start of the controller, the motors start and never stop, moving the gondola out of frame. 

To replicate the error:
1) connect device (ramps in my case)
2) open the controller
3) press any arrow or drag the gondola

To avoid the error:
1) connect device (ramps in my case)
2) open the controller
3) set home
4) press any arrow or drag the gondola

The gondola moves as expected.

RickMcConney, 2015-09-05:
Not sure why this is happening. I send the home command at startup. The arrow keys also send a command to put the ramps into relative mode then sends a movement command of 1 mm. I think the problem is that the ramps is missing the machine spec command so the steps per rev is unset. I noticed that I did not see the print out of pen width set when sending the initial machine spec. So I now send that command first. This may fix the issue. Check it out in V08 and let me know if it is fixed.

patriciorivera, 2015-09-08:
Fixed.

closed 2015-09-08 by RickMcConney