RickMcConney/PenPlotter #5: gcode type

mcmasterp, 2016-05-07:
Not sure if this is the right way to go about this but I am using vetrics aspire to generate toolpaths. it has many post processors to output gcode into tons of formats for different machines. I tried a few that seemed generic bt when I load the resulting .nc files it's blank and clicking plot does nothing. thanks for the awesome software though!

RickMcConney, 2016-05-09:
The gcode parser is pretty simplistic try the grbl post processor if Vetrics has one that should be the closest. Basically you want to end up with a file that just has G0, G1, G2, G3, G21 and G20 commands in it, the parser should skip over other lines. This will cause problems with some post processors that only output the G1 or G0 command once then just have X, Y lines. Each line must start with a G command.

mcmasterp, 2016-05-09:
Oh yes I should have replied. It does work perfectly, it was my mistake. I
size the work piece in ventric to be the size of paper I used in inches. I
had to change it to mm and make the piece quadruple the size. I finally
noticed this when I realized that loading geode created a tiny dot on the
page. Sure enough after a whole lot of zooming in it was there.

Thanks again for the amazing work. I took my drawbot to mothers day brunch
and drew happy mothers day signs for all the moms. Everyone loved it

closed 2016-08-24 by RickMcConney