RickMcConney/PenPlotter #12: Servo range not defined by default.txt?

kongorillaSJK, 2017-07-07:
Changes to the up/down servo values in the default.txt aren't having an effect for me. Do I have to change the values in the firmware? If so, can you give me a pointer? (I did do a quick search through the various files of the firmware, but it wasn't apparent where I should make the change).
I've adjusted other values in the txt and those had an effect, so I assume I'm not doing something too stupid.
TIA

RickMcConney, 2017-07-10:
Thanks you are correct I was not using the values read from the config file. I have made the correction. It is in the sendPenUp and sendPenDown functions in the comm.pde file. Let me know if this fixes your issue.
Rick/

kongorillaSJK, 2017-09-27:
Thanks, Rick! This indeed fixed the issue. (Sorry for the late reply; time for plotting has been hard to come by lately).

closed 2017-09-27 by kongorillaSJK