RickMcConney/PenPlotter #13: lcd support

ekaggrat, 2017-12-04:
great work.. i had made the first plotter using the link you had provided but it became defunct as it always needed a laptop to print.. 

does your version of repetier support printing from sd card and lcd?

thanks

RickMcConney, 2017-12-04:
I have not tried it myself but it should work. You may need to set the LCD in the config file to match the one you have.

gehringer, 2017-12-04:
Printing from SD-card worked for me using an old Arduino mega with RAMPS 1.4 and repetier... setting up repetier just took some time and changing paramaters until I found the right parameters for my hardware.
Thanks for providing the instructions Rick!

robscar, 2017-12-13:
just got the lcd (reprapdiscount full graphics controller) working. Edit FEATURE_CONTROLLER to fit your display and make sure you use 1.05-r2 arduino environment for compiling. I ran into strange compile issues with the newest version.

mshaub, 2018-03-25:
Does anyone know if this will work with the standard Polargraph hardware (Arduino Mega and Polarshield motor drivers)? I'd love to try this out as an alternative but not sure how to start. Thanks for any advice you can provide!

RickMcConney, 2018-03-26:
It uses marlin on the Arduino Mega so no problem there but I am not sure marlin supports the polarshield. I would just get a ramps board they are less than $10 on Amazon.