RickMcConney/PenPlotter #14: Needs a button toggle to suppress the image while drawing.

Shornone, 2018-02-12:
I think it is just a resource thing having to render the image and draw the command trace bogs processing down. Framerates can get down to 1-3 fps which is just too slow for smooth drawing as it waits to update the command queue until after the rending update is complete.

A toggle button like the pen up/down feature would free up the processor.

Shornone, 2018-03-19:
Just a follow-up. I have been running this on a Raspberry Pi 3 B and the constant screen updates for larger vector files just eats up the processor and eventually everything grinds to a halt. When I resize the PenPlotter controller so that the image is not visible I get at least a 2x increase in drawing speed, but it doesn't hold true from start to finish. Based on this experiment I believe being able to suppress the loaded vector image and only displaying the drawing progress would help immensely.

 I have been able to add the button, but not being very good with the software I can't quite seem to get the function correct to hide the original image and only show the plotting updates.

[image]

https://twitter.com/ShornOne/status/975131151563018240
https://twitter.com/ShornOne/status/975130168103641089

Shornone, 2018-03-22:
Perhaps a standalone command queue sender running in Python is the ticket. It really is a great controller having done about 20 drawings,  if I can get the resource need down for the raspberry pi it will be golden. Limited to simple line drawings for the time being.

Thanks

-Shorn

RickMcConney, 2018-03-23:
Unfortunately the processing programming environment is designed to update the screen in a continuous loop. Now it may still be possible to optimize the code or skip some drawing I am not sure this effort would be worth while on a Pi. You may instead be able to run the program on a laptop and export the gcode then use something like octopi on the pi to send the gcode to the plotter. Your plots look great.

Shornone, 2018-03-23:
Thanks for the feedback. Sandy Noble's Polargraph controller has a toggle for suppressing the original vector which is what I was inquiring about to replicate in your controller. Not the generated plotting code overlay or actual drawing in progress feedback.

Also, I was able to modify Sandy's standalone command line sender to talk to the Arduino and send commands but something is still amiss. 

The Pi is desirable because I set up the machine in art galleries and leaving a laptop tethered is risky.

Thanks again for responding. I will check out octopi. I think getting a command line sender is probably the best for the Pi. If I make any progress I will share back as a courtesy.

Thanks again for taking the time! - Shorn

Another example of the more artistic drawings I do with the machines - 1933 Plymouth Racer

[image]

RickMcConney, 2018-03-23:
I added a feature that you can try to see if it solves your speed issue. If you hit the 'n' key it will go into a nodraw mode hit the 'd' key to resume drawing. All the changes are in the main PenPlotter.pde file. The routine to look at is called draw I have added a boolean called draw that if false will skip the drawing calls. At the moment all drawing is skipped. Let me know if this helps.

Shornone, 2018-03-23:
Sweet! When I get home (... I...should... be working right now...) I will definitely give it a try first thing. Thanks!

Shornone, 2018-03-23:
Works Great!

~21 fps with no file loaded
~4 fps with file loaded
And a solid 30 fps with draw suppressed.

Much much thanks!

I offered on Twitter a CanaKit Raspberry Pi 3 B+.

https://twitter.com/ShornOne/status/976981890891526144?s=19

It is yours if you want it or I can donate to the local high school.
[image]
[image]
[image]

Thanks you again and let me know about the CanaKit.

-Shorn

Shornone, 2018-03-23:
Ran the stress test and it worked like a dream. Held 30+ fps for the entire drawing. Draw time under 2hrs. Couldn't even finish the drawing before.

[image]

Thanks again!

RickMcConney, 2018-03-23:
Glad it worked for you sorry I did not notice your post sooner. Thanks for the offer of the Pi but I do not need one I am swimming in them here at work.

jsecondo, 2018-04-17:
Wow. That stress image is amazing. Can you share it! Thanks

Shornone, 2018-04-17:
The SVG file I used came straight from Wikipedia. Smith Chart SVG. I just downloaded it, scaled it and then printed it.

https://commons.m.wikimedia.org/wiki/File:Smith_chart_gen.svg

mshaub, 2018-04-18:
Yes, thanks for the link!
I just ran the same Smith Chart file (without pen up/down working) and didn't have quite as beautiful results. It looks a little like when I try to write with my left hand ;) but certainly better than that. Any guesses about what settings need to be tweaked to smooth this out? The drawing is about 16" in diameter.
https://www.dropbox.com/s/3fmfm4ndcucpul7/Photo%20Apr%2018%2C%208%2005%2030%20AM.jpg?dl=0

Shornone, 2018-04-18:
Without more information about your setup, it is hard to diagnose. From my experience, most issues like yours are related to the gondola/pen holder but I just can't say for sure without more details of your system.

On my twitter account, you can see my setup in operation. I have videos and images that you can reference.

https://twitter.com/ShornOne