RickMcConney/PenPlotter #15: Is this possible/worth porting this to ESP8266 with custom DRV8825 carrier shield?

eimix, 2018-03-29:
I'm going to build "vertical plotter", and want to make it WiFi way (ESP8266).

Did not found yet project what already did it.

Your firmware with **gcode** looks more usefull and promising then "polargraph's".
It's possible to use arduino with WiFi module, but ESP8266 could do everything.

Maybe you (this project builders) could tell if is possible, or if there is any obvious problems not to do it WiFi way?

RickMcConney, 2018-03-29:
The part I did was just the GUI with small changes in Marlin and repetier. It is marlin or repetier that would need to be ported. That is a big job. But instead you may be able to use a raspberry pi with octoprint. It is the standard way to add wifi support to a 3d printer running marlin. You would use my software to generate the Gcode then upload it to the pi which directly controls marlin.

eimix, 2018-03-29:
Thanks, I will try look it that way.

Tried penPlotter_win32.zip app v08
from
https://github.com/RickMcConney/PenPlotter/releases

On Windows 7 PC, javaw.exe shows up in taskmanager, but nothing is on display.
Tested two separate PC - same result. (both Windows7)

Any other software (Java runtime environment, JDK) should be installed to run it?

RickMcConney, 2018-03-29:
Not sure why that is happening. You can just download the project source code and run it with processing. You load the main file PenPlotter.pde into processing and then hit the run button.