RickMcConney/PenPlotter #17: Pen Up/Down

mshaub, 2018-04-18:
More questions from a n00b, sorry. There are references to servo settings in the processing sketch and in the RepRap firmware, but it's not obvious to me how to hook up the servo to my RAMPS

RickMcConney, 2018-04-18:
The servo is used to drive a servo that should lift and lower the pen.  You need to make sure the servo pins are powered as by default they are not. To power them you can jumper the vcc and 5v pin located near the reset button on the board. The software uses servo 3 which is the farthest set of pins in servo block from the reset button. You can then click on the pen up and pen down buttons in the gui to test the servo.

You can adjust how far up and down the servo moves with the values in the default.properties.txt file the default values are

servo.downValue=1500
servo.dwell=250
servo.upValue=2350

mshaub, 2018-04-18:
Thanks Rick! The jumper is the missing piece for me. Will try tonight!

mshaub, 2018-04-19:
Got it working, thank you! 
The jumper location is completely different on RAMPS 1.6
Here are a couple marked up images of where I found it.

https://www.dropbox.com/s/ey4xyiniqfruau2/Photo%20Apr%2018%2C%2010%2002%2004%20PM.pdf?dl=0
https://www.dropbox.com/s/jl5wj0tia8lzs4y/Photo%20Feb%2006%2C%202%2047%2026%20PM.pdf?dl=0

venuasg, 2020-09-24:
> Got it working, thank you!
> The jumper location is completely different on RAMPS 1.6
> Here are a couple marked up images of where I found it.
> 
> [image]
> [image]

When i add the jumper, board did not start and when removed it worked again, adding jumper is making board to sleep, what is wrong. Pls help. I am trying to connect BLTOUCH without luck.

venuasg, 2020-09-26:
Hi, Finally i was able to solve with alternate approach. 5v pins on Servo on RAMPS 1.6 does not work and i had to connect it to AUX 1 5v and BLTOUCH started working....

maartenvant, 2021-08-04:
adding the jumper to my RAMPS 1.6 Plus Board it blew up my Arduino..... so I guess the jumper is not enabling the 5v on the Servo's.... anybody got an idea how to enable the 5v on the servo pin part? Gets a bit messy if need to split the cables and get the 5v from somewhere else on the board....