RickMcConney/PenPlotter #18: Polargraph / CoreXY

jsecondo, 2018-04-19:
Rick,

I made a polargrah like  Sandy Nobles and a corexy like Axidraw.

Both can be controlled by your controller. Its clear that both Repetier and Marlin support them. But how about the penplotter controller?

RickMcConney, 2018-04-19:
The penplotter controller communicates with my modified version of Marlin and Repetier. There are a couple of custom gcodes and the polar coordinate support needed for the polargraph machine. However the gcode can be saved and used on any machine that supports gcode.

jsecondo, 2018-04-19:
What happens if I connect a COREXY to the PenPlotter Controller?.

Basically the info transmitted to the machine are GCODES. For example, if I
load an image and apply some effect, the GCODE generated is generic gcode
(not polargraph specific gcode), so if the machine accepts gcode from the
serial it might work pretty straight forward.

Or am I missing something???

RickMcConney, 2018-04-19:
That's correct. But do not use my version of Marlin use the version that works with your corexy machine as I override the steps per mm. The Diamond treatment will not work as that uses custom gcode. Also be careful of the position of you file on the pen plotter page. The X is normally about 450 mm and the Y is 250 mm which may be larger than your machine. You can drag the plot with the right mouse button up to the top left corner of the page which is 0,0. Export and check the gcode with an editor or by loading it into something like cura or repetier before running it on your machine or at least be near the power switch on your first run.

jsecondo, 2018-04-20:
Great!!
I'll try it this weekend.

Thanks a lot.

By the way. I'd like to make an automatic changing pen gondola for the
vertical plotter. I've already programmed a cmyk color separation routine.
I want to add this feature to your controller. Like a checkbox to
 separation and run the different drawing styles on each component image
and create a single gcode file with changing pen commands in the middle.
Any suggestions before throwing myself intoncoding this? You know the soft
better than anyone.

Another thing running through my mind was to replicate the spiral style in
polargraph.

jsecondo, 2018-04-25:
> I had another user that did a similar thing. They used photoshop to do the
> color separation then ran each separation through the controller and
> manually changed the pen between each run, the result was quite good. As
> the runs take so long there was not really a need to automate the two pen
> changes. But feel free to try. You can add a new servo command to control
> the pen changing servo with out having to modify the marlin code just add a
> new command in the comm.pde file see how the pen up and down are
> implemented.

Already on it. Great coding by the way.