RickMcConney/PenPlotter #20: Export Options

TomSoto, 2018-06-08:
Hi, I m just testing the awesome code you have created. 
Is there any way, to export as any other vector based format like eps, pdf, svg or dxf rather than only gcode ?
Best regard from Germany
Tom

RickMcConney, 2018-06-08:
Sorry it only exports gcode. But if you do a google search you can find converters that can convert gcode to dxf. Or if you are handy at coding you could modify the export.pde to support your preferred format.

TomSoto, 2018-06-08:
Hi, thanks a lot for the tips ! I ll have a look on the internet.
Best,
Tom

kongorillaSJK, 2018-06-08:
I've used http://scorchworks.com/Gcoderipper/gcoderipper.html for this.

IIRC, every line is a separate segment, rather than connecting coincident points, which isn't optimal or convenient.

TomSoto, 2018-06-09:
many thanks ! that looks good indeed !!