RickMcConney/PenPlotter #21: Unable to compile

pearlson, 2018-11-05:
Hi, 

My environment is as following 
Arduino IDE : 1.8.7 (I have also tried 1.8.6 and 1.8.5 without any success)
Downloaded latest version of penplotter 0.8 
I have GRBL added in libraries. But I guess that should not make any difference in compiling Repetier
I have selected Arduino Mega 2560 in Arduino IDE for compilation 

Compilation fails with following errors

Arduino: 1.8.7 (Linux), Board: "Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"

In file included from sketch/HAL.h:77:0,
                 from sketch/Repetier.h:264,
                 from /home/muthu/Downloads/penplotter-0.8/PenPlotter-08/Repetier/Repetier.ino:127:
pins.h:323:36: error: pasting "/*38*/" and "_WPORT" does not give a valid preprocessing token
 #define ORIG_X_ENABLE_PIN       30 //38
                                    ^
sketch/fastio.h:29:47: note: in definition of macro '_WRITE'
 #define  _WRITE(IO, v)   do { if (v) {DIO ##  IO ## _WPORT |= MASK(DIO ## IO ## _PIN); } else {DIO ##  IO ## _WPORT &= ~MASK(DIO ## IO ## _PIN); }; } while (0)
                                               ^
sketch/Printer.h:206:9: note: in expansion of macro 'WRITE'
         WRITE(X_ENABLE_PIN,!X_ENABLE_ON);
         ^
sketch/pins.h:2214:22: note: in expansion of macro 'ORIG_X_ENABLE_PIN'
 #define X_ENABLE_PIN ORIG_X_ENABLE_PIN
                      ^
sketch/Printer.h:206:15: note: in expansion of macro 'X_ENABLE_PIN'
         WRITE(X_ENABLE_PIN,!X_ENABLE_ON);
               ^
pins.h:323:36: error: pasting "/*38*/" and "_PIN" does not give a valid preprocessing token
 #define ORIG_X_ENABLE_PIN       30 //38
                                    ^
sketch/fastio.h:29:75: note: in definition of macro '_WRITE'
 #define  _WRITE(IO, v)   do { if (v) {DIO ##  IO ## _WPORT |= MASK(DIO ## IO ## _PIN); } else {DIO ##  IO ## _WPORT &= ~MASK(DIO ## IO ## _PIN); }; } while (0)
                                                                           ^
sketch/Printer.h:206:9: note: in expansion of macro 'WRITE'
         WRITE(X_ENABLE_PIN,!X_ENABLE_ON);
         ^
sketch/pins.h:2214:22: note: in expansion of macro 'ORIG_X_ENABLE_PIN'
 #define X_ENABLE_PIN ORIG_X_ENABLE_PIN
                      ^
sketch/Printer.h:206:15: note: in expansion of macro 'X_ENABLE_PIN'
         WRITE(X_ENABLE_PIN,!X_ENABLE_ON);
               ^
pins.h:323:36: error: pasting "/*38*/" and "_WPORT" does not give a valid preprocessing token
 #define ORIG_X_ENABLE_PIN       30 //38
                                    ^
sketch/fastio.h:29:104: note: in definition of macro '_WRITE'
 #define  _WRITE(IO, v)   do { if (v) {DIO ##  IO ## _WPORT |= MASK(DIO ## IO ## _PIN); } else {DIO ##  IO ## _WPORT &= ~MASK(DIO ## IO ## _PIN); }; } while (0)
                                                                                                        ^
sketch/Printer.h:206:9: note: in expansion of macro 'WRITE'
         WRITE(X_ENABLE_PIN,!X_ENABLE_ON);
         ^
sketch/pins.h:2214:22: note: in expansion of macro 'ORIG_X_ENABLE_PIN'
 #define X_ENABLE_PIN ORIG_X_ENABLE_PIN
                      ^
sketch/Printer.h:206:15: note: in expansion of macro 'X_ENABLE_PIN'
         WRITE(X_ENABLE_PIN,!X_ENABLE_ON);
               ^
In file included from sketch/HAL.h:77:0,
                 from sketch/Repetier.h:264,
                 from /home/muthu/Downloads/penplotter-0.8/PenPlotter-08/Repetier/Repetier.ino:127:
pins.h:323:36: error: pasting "/*38*/" and "_PIN" does not give a valid preprocessing token
 #define ORIG_X_ENABLE_PIN       30 //38
                                    ^
sketch/fastio.h:29:133: note: in definition of macro '_WRITE'
 #define  _WRITE(IO, v)   do { if (v) {DIO ##  IO ## _WPORT |= MASK(DIO ## IO ## _PIN); } else {DIO ##  IO ## _WPORT &= ~MASK(DIO ## IO ## _PIN); }; } while (0)
                                                                                                                                     ^
sketch/Printer.h:206:9: note: in expansion of macro 'WRITE'
[... 354 lines cut ...]
 #define ORIG_X_STEP_PIN         36 //54
                                    ^
sketch/fastio.h:29:133: note: in definition of macro '_WRITE'
 #define  _WRITE(IO, v)   do { if (v) {DIO ##  IO ## _WPORT |= MASK(DIO ## IO ## _PIN); } else {DIO ##  IO ## _WPORT &= ~MASK(DIO ## IO ## _PIN); }; } while (0)
                                                                                                                                     ^
sketch/motion.h:501:9: note: in expansion of macro 'WRITE'
         WRITE(X_STEP_PIN,HIGH);
         ^
sketch/pins.h:2212:20: note: in expansion of macro 'ORIG_X_STEP_PIN'
 #define X_STEP_PIN ORIG_X_STEP_PIN
                    ^
sketch/motion.h:501:15: note: in expansion of macro 'X_STEP_PIN'
         WRITE(X_STEP_PIN,HIGH);
               ^
exit status 1
pasting "/*38*/" and "_WPORT" does not give a valid preprocessing token

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

Any help appreciated 

Regards
Muthu

per1234, 2018-11-06:
It's strange because it sure looks like https://github.com/arduino/Arduino/issues/7930, which was supposed to be fixed with https://github.com/arduino/arduino-builder/pull/285. But that change was released in Arduino IDE 1.8.7.

pearlson, 2018-11-06:
Just checked older issues . E1 maps to left motor and Y maps to right one. Will try that out. 

Thanks again. 

If it's OK, this issue may be closed

Regards
Muthu

pearlson, 2018-11-06:
Hi, 

The whole setup (without servo) is working perfectly. The movements are fine and I think I would be able to do some good work. Since I am familiar with GCodes and I was looking for something compatible, Penplotter firmware and controller looks to be apt for me. 

Next step is to try some drawings. Its late night in this part of globe and I will get time only by next weekend. 

Thanks for a wonderful  work again. 

I will close the current issue. 

Regards
Muthu

closed 2018-11-06 by pearlson