RickMcConney/PenPlotter #28: (FR) Adapiting for use with GRBL.

ithinkido, 2019-07-12:
Hello Rick,

Do you think it would be possible to add the plotter geometry calculations to GRBL so that it would be possible to use this as a plotter firmware.
The motivation for this is that there are some very nice projects coming along that are running GRBL on boards like the ESP32 and some laser plotters that even use just an ESP8266 ( issue #15). GRBL is lighter that Marlin or Repetier and could also be run on the UNO or other such boards that can not manage Repetier or Marlin.
It seems to be a good match to the needs of the plotter. We do not need heater mosfets or temp control, so using boards like the RAMPS and a MEGA is a bit of an overkill, and although they are not the most expensive things, an UNO and a CNC shield are even cheaper, not to mention the possibility of running it on an ESP32 which open up even more great possibilities like wireless control.
As far as I know the original Polargraph was also able to run on these platforms, and I think it would be amazing to see this possibility too with your version.

RickMcConney, 2019-07-17:
Sorry I no longer have the setup to test out such a port. But The changes to the code are flagged with a POLAR define if you want to try the port yourself. You would need to find the equivalent places in GRBL to add the changes.

ithinkido, 2019-07-17:
Thanks for the reply. I started looking through the Marlin code at that and I will keep you updated with the progress. In the mean time I will close this issue

closed 2019-07-17 by ithinkido