RickMcConney/PenPlotter #30: Switching Z Servo to Stepper

akzmonster, 2019-09-05:
So, first off, this is awesome! It would be nice if I could implement end-stops so that it could auto home on startup every time, but that's not a big deal right now.

What I would really like to do is switch the Z servo for the Z stepper. Is there anything I have to do or should I just plug in the Z motor and go?

RickMcConney, 2019-09-06:
You will need to modify the sendPenUp and sendPenDown routines in the comm.pde file to send G0 Z commands instead of servo commands. An example can be seen in the routines of the same name in the export.pde file. It will use the go Z gcodes if the export file ends with .cnc

akzmonster, 2019-09-06:
Awesome, thanks! I'll check it out later today.