RickMcConney/PenPlotter #31: Doesn't compile on any Arduino IDE version for me

Ohmarinus, 2020-02-19:
What ever I try, I cannot get the firmware to compile.
I tried both old and new versions of Arduino IDE.

This is the error I'm getting in Arduino IDE 1.8.9

`/var/folders/bq/rbn1n5x90g3fj8dt7cskqzxm0000gn/T/arduino_build_302033/sketch/ui.cpp: In member function 'void UIDisplay::adjustMenuPos()':
ui.cpp:2016:41: error: expected primary-expression before 'const'
     uint8_t mtype = HAL::readFlashByte((const prog_char*)&(men->menuType));
                                         ^
ui.cpp:2016:41: error: expected ')' before 'const'
/var/folders/bq/rbn1n5x90g3fj8dt7cskqzxm0000gn/T/arduino_build_302033/sketch/ui.cpp: In member function 'void UIDisplay::nextPreviousAction(int8_t)':
ui.cpp:2083:41: error: expected primary-expression before 'const'
     uint8_t mtype = HAL::readFlashByte((const prog_char*)&(men->menuType));
                                         ^
ui.cpp:2083:41: error: expected ')' before 'const'
ui.cpp:2087:43: error: expected primary-expression before 'const'
     uint8_t entType = HAL::readFlashByte((const prog_char*)&(ent->menuType)); // 0 = Info, 1 = Headline, 2 = submenu ref, 3 = direct action command
                                           ^
ui.cpp:2087:43: error: expected ')' before 'const'
Using library SPI at version 1.0 in folder: /Users/marinusdebeer/Library/Arduino15/packages/arduino/hardware/avr/1.6.21/libraries/SPI 
exit status 1
expected primary-expression before 'const'
`

I'm not sure what the issue is. On different versions of Arduino IDE I get different errors. For example with 1.6.9 I get this error:
**fatal error: twi.c: No such file or directory**

I can compile Marlin etc without a problem.

Would anyone have a fix?

RickMcConney, 2020-02-25:
Not sure what the issue is I am using version 1.8.9 of the IDE and I can compile both the marling and repetier. The second error you show fails to find the file twi.c it is located in the ./Repetier/libraries/twi.c. However marlin can be used you do not need repetier either will work. Have you set the board type in the tools menu to arduino mega?

Ohmarinus, 2020-02-27:
Hello Rick, thanks for your reply! I would really like to try out your Repetier plotter firmware though.

Currently I'm using Makelangelo which works fine, however I want to get familiar with more different firmwares and decide which one I like best.

I am able to compile Marlin and Makelangelo. It's only this package that is causing issues for me. The board was set to the right one and I've also set the correct baud rate. Uploading won't be the issue I think, it's just that the firmware won't compile with 'Verify'. It's kind of a persistent error.

Haven't really found a solution yet so for now I'll work with Makelangelo firmware, but would really like to give yours a try. My CoreXY plotter is using a self-hacked Marlin version and I dislike Marlin for pen plotters.