RickMcConney/PenPlotter #33: Inconsistent behaviour between gcode sent from controller and exported gcode

SimonArnu, 2020-11-14:
Hi Rick,
first things first: I really like your software, it's really a step up from the standard polargraph software stack.
I think the concept of using mostly standard gcode is really useful as it can decouple creation and execution of instructions as I am used from my 3D printer (e..g. utilising octoprint).
But maybe I am misinterpreting the whole setup.
My assumption is: I can place some svg in the controller, adapt paper size, scale and placement of drawing and hit 'export'.
Then I can take the generated gcode and send it to the plotter using some simple utility/script that sends it over the serial port.
Then the plotter plots happily until the end of the file. This way, I can attach a raspberry or an old laptop to the plotter without tying my current laptop to the machine for hours.
I have an issue with this kind usage, though. The generated gcode seems to be off by about half of the drawing area, which sends the gondola out of the bounds when starting the plot.
Plotting directly from the controller works as it should.
I could use some tool to modify the gcode after exporting to change that offset, but I assumed that was not the way it is intended to work, right?
Am I missing something or is there a bug in the export?

RickMcConney, 2020-11-16:
I think the issue you are seeing is that the gcode generated is relative to the origin of the plotter which is the center of the gondola after it has been homed. The exported gcode does not include commands to set the home as it exported and can be run on any plotter. So I think you need to include some gcode commands to set the home position or do this manually before using the exported file. If you are plotting to the same plotter that uses my version of marlin then you will also need  the following M4 command to set up the firmware parameters. 
M4 <machineWidthInMM> E <penWidthInMM> S <stepsPerRev> P <mmPerRev>

You may also be able to connect to the plotter and do a home using my GUI. Then disconnect and use the exported file from octoprint.

I have not tried but I think the code should be able to be run on a raspberry pi and then you could just use the GUI from the pi.

SimonArnu, 2020-11-17:
Hi Rick,
thanks for the answer. I was using the Repetier version, but will give Marlin a try.
And prepend my gcode with commands for homing.