RickMcConney/PenPlotter #38: NOT a polar pen plotter Is a wall based pen plotter

WalterHynson, 2021-11-27:
How or why these things are called polar pen plotters is beyond me as there is no rotating theta axis,It is a X,Y based pen plotter and the name polar leads to identification issues.

kongorillaSJK, 2021-11-27:
In answer to the "how or why," I'll quote from Sandy Noble's original polargraph site.

> "It’s called a polargraph because when I was making it, I thought of it as using a dual-polar coordinates system internally, rather than the regular cartesian system we (and computer systems) tend to use.
> 
> In fact, it doesn’t use a polar system at all, it’s actually a kind of double-triangulation coordinate system. The angle of each cord is controlled by the length of both cords, rather than by specifying angle and distance as in with a true polar coordinate. The name “Polargraph” remains as an evocative term, but it isn’t accurate."

http://www.polargraph.co.uk/whats-a-polargraph/

kongorillaSJK, 2021-11-27:
Oh, forgot to add, from the same page of the polargraph site, Sandy Noble writes:

> **What’s in a name?**
> 
> > Polargraph is a particular brand of drawing machine. It is often used in a generic way to describe any of these kinds of machine, and I take that as a compliment to the ubiquity of the project, but it shouldn’t really be used except when referring to this Polargraph project.
> > 
> > It is not a generic name for a drawbot, and the term was not in use before this project began. Well, you know, except to refer to polar graphs.

closed 2025-05-10 by WalterHynson